kriswiner/CMWX1ZZABZ #1: images

kriswiner, 2018-01-13:
[image]

kriswiner, 2018-01-15:
[image]

kriswiner, 2018-01-17:
[image]

kriswiner, 2018-01-19:
[image]

kriswiner, 2018-02-09:
[image]

kriswiner, 2018-04-06:
> Hi Kris,
>
> Lora is a little bit confuse for me at the moment.
> So here, some questions :
>
> Could I use the card in France? using the Europe Frequency (EU868)?
> How to get data that I send? Does I need to find a provider in France?
> Does I can use the https://www.thethingsnetwork.org/ services?
> Does I pay to a mobile provider to use antenna because data are sent using
> the mobile network?
>
> Thanks,
> Cédric

Hi Cedric,

Grasshopper (and the other CMWX1ZZABZ products) is designed to work at
European frequencies.

Typically, the Grasshopper would be used with a LoRaWAN gateway. I have
both the Multitech AP (which I would recommend) and the Things Network
gateways. Both work well enough although the TTN gateway is having some
problems at the moment. The Multitech AP is ~$300.

The Things Network is a network of privately owned gateways. In the major
cities in Europe the coverage is sufficient that you don;t need your own
gateway, some TTN agateway will capture your data. Look up the local TTN
members in your area to see what the coverage is. I am sure in Paris you
wouldn't need a gateway.

The gateway receives the LoRaWAN data from the Grasshopper, and then
forward the data packets to a LoRa server. I use the SEMTECH server but
there are many available at no cost.

Then the LoRa server pushes the data to whatever back end you want; I use
myDevices and CayenneLPP.

The things network has a lot of tutorial information so look there, they
also have a forum where you can ask more detailed questions.

You shouldn;t need a mobile network subscription.

Kris

kriswiner, 2018-07-14:
[image]

kriswiner, 2018-10-24:
[image]
[image]
[image]

kriswiner, 2018-12-19:
[image]

kriswiner, 2018-12-19:
[image]

kaebmoo, 2018-12-20:
New version?

kriswiner, 2018-12-20:
What is your question?

kriswiner, 2019-07-03:
[image]

kriswiner, 2019-07-03:
[image]

kriswiner, 2022-08-11:
[image]

kriswiner, 2022-08-15:
[image]

kriswiner, 2022-08-15:
[image]

kriswiner, 2022-11-15:
[image]

kriswiner, 2023-01-01:
[image]

kriswiner, 2023-01-01:
[image]

kriswiner, 2023-01-22:
[image]

kriswiner, 2023-01-22:
[image]

kriswiner, 2023-01-22:
[image]

kriswiner, 2023-01-22:
[image]

kriswiner, 2023-01-22:
[image]