kriswiner/CMWX1ZZABZ #5: LoRaSensorTile.v05 hangs after a few iterations

ssozonoff, 2018-04-05:
Hello Kris,

I have been testing LoRaSensorTile.v05 example by simply changing the Lora frequency for EU and the appEui, appKey and devEui.

I am unable to keep things running for like more than 3 iterations max. Then the Tile hangs or crashes. Does not wakeup with movement or anything and requires a reboot to get going again. My gut feeling is that its something on the Lora side but I need to do some more testing. 
Have you had this code running for more than a few hours ? If so then I dont understand.

Thanks,
Serge

ssozonoff, 2018-04-05:
OK so I commented out the Lora begin, write and end stuff but still crashing so its not that .... We start to review the code as soon as I get a chance. At this stage impossible to say if its a bug in the Core or the sketch.

ssozonoff, 2018-04-05:
So on a hunch I ripped out all the code related to the BMA chip including the interrupts. Its now been running for an hour...

kriswiner, 2018-04-05:
Hi Serge,

Maybe try at 4.2 or 16 MHz clock speed. The USB serial might be blocking. I
thought this was resolved in the latest core but maybe it has returned?

In any case, I have been running several LoRaSensorTiles continuously for
more than a month.

kriswiner, 2018-04-05:
This shouldn't have anything to do with the BMA280.

Are you talking about the Cricket asset tracker or LoRaSensor tile?

You need to be careful to select the Cricket board for the Cricket asset
tracker and the Cicada board for the LoRaSensorTile.

ssozonoff, 2018-04-05:
I am using the LoRaSensor tile. I will try selecting the Cicada board and see if that makes any difference.

kriswiner, 2018-04-05:
Also try at 4.2 or 16 MHz clock speed.

ssozonoff, 2018-04-05:
Currently testing at 16Mhz and Cicada board. Will keep you posted. When I say crashes I mean crashes after a certain amount of time typically 10 to 15 minutes.

kriswiner, 2018-04-05:
Been running for twenty minutes at 32 MHz, I'll keep reporting for a while,
but I am sure this will go for months before "crashing"...

kriswiner, 2018-04-05:
OK, been running for 70 minutes now, I think I will declare victory...

ssozonoff, 2018-04-05:
Hi Kris,

At 32 I have not been able to make it run longet than 15min, at 16Mhz its been running for over an hour.

[image]

kriswiner, 2018-04-05:
No idea, I just ran one of mine at 32 MHz for more than an hour. Of course,
I am not using TTN. Not sure this should matter.

I assume you are using a battery? or are you connected to USB the whole
time?

Maybe your serial monitor times out???

ssozonoff, 2018-04-05:
Well not only dont I have any serial output but I also dont receive any data to TTN at the point where things appear to crash. I will keep experimenting and see if I can get more detail. Thanks

I have been using both a batter and cable connection to computer USB port.

kriswiner, 2018-04-06:
> Not a TTN issue. Mine has run all night at 16Mhz however at 16Mhz their is
> no Serial Monitor so the only thing I can think of is running at 32Mhz and
> the Serial Monitor open could be causing something. I am going to retry
> today at 32Mhz again.

There was an issue a few months back where the sketch would stall as the
serial buffer filled up if the serial monitor was not opened. This was
fixed though. Also, there is an VBAT call that can stall the program if it
is placed near the end of the llop but not at the top of the loop. Still
don;t understand this one.

But if you use the latest Ardiono core and choose the Cicada board and use
the sketch I sent you, it should work without any trouble. I just tested
this today at 32 MHz and it ran for 70 minutes withno trouble...

ssozonoff, 2018-04-06:
Now I just need to figure out how to get the USB port back since it appears I lost it when switching to 16Mhz .... Trying instructions from the end of the Core Readme.....

kriswiner, 2018-04-06:
Press and hold boot button, press and release reset. Flash a program at 32
MHz will return the COM port. You might have to restart the Arduino IDE if
there is an old COM port there.

ssozonoff, 2018-04-06:
Hi Kris,

So seems to run in any mode as long as I dont have a Serial Monitor attached.... I can confirm that I am up to date with the Arduino Core. I did a Git pull this morning.

kriswiner, 2018-04-06:
Hi Serge, I have been running for more than 30 minutes using a
LoRaSensorTile at 32 MHz with the serial monitor open and I am still
getting LoRaWAN updates every ten minutes and serial updates every minute
as expected.

Perhaps the problem you are seeing is with a bad USB cable/bad USB
connection such that the serial stops or stops intermittently and causes
your board to "crash".

I can't reproduce your problem here. Please try using a different
cable/computer and see if the problem recurs.

ssozonoff, 2018-04-06:
Well mine has run all day without Serial Monitor. I am of course happy to try with a different USB cable however my Mac USB has never failed me with development around other Arduino boards and projects so I dont expect it to start now.

kriswiner, 2018-04-06:
Not sure what to say, I just ran for an hour at 32 MHz with serial and
getting LoRaWAN data to the SEMTECH server, no problem. I can't reproduce
your "error" so it's hard to say whether this could be a software or
hardware problem, but I don't see it so I am somewhat at a loss to
recommend a course of action. I am using a Windows 10 machine so maybe
there is something quirky about a MAC and the serial port on the
LoRaSensorTile. Not sure what it could be. Maybe you could try it on one of
the Grasshoppers you'll be getting and see if it happens there too?

ssozonoff, 2018-04-06:
Hi Kirs,

Not worries I know now at least more or less what seems to cause the issue and in the "real world" there will be no serial monitor. I think it is good to have this documented here in case it rears its head again with some else.

Thanks for your help.
Serge

kriswiner, 2018-04-06:
What is the cause? I can't seem to reproduce it.

ssozonoff, 2018-04-06:
Well for me it seems to be serial monitor related since its running fine at 32Mhz on battery. I will still test with another cable and see. I will also retest a couple of more times with the Serial Monitor to see if the behavior is consistent as well.

kriswiner, 2018-04-06:
Yes, thank you, if it is a flaw in the hardware or Arduino core, of course,
we would like to know. Thanks for testing!