kriswiner/CMWX1ZZABZ #6: STM32L0.sleep() and STM32L0.stop() not working on B-L072Z-LRWAN1

jaimelaborda, 2018-06-22:
I'm using a B-L072Z-LRWAN1 but methods sleep() and stop() of the STM32L0 library appears to be not working. It compiles without error, but the programs just runs no matter what value you put into the methods.

My code:

#include <STM32L0.h>
#include <Adafruit_SHT31.h>
#include <Adafruit_TSL2591.h>
#include "LoRaWAN.h"

const char *appEui = "XXXXXXXXXXXXXXXX";
const char *appKey = "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX";
const char *devEui = "XXXXXXXXXXXXXXXX";

#define LPP_DATATYPE_TEMPERATURE 0x67
#define LPP_DATATYPE_HUMIDITY 0x68
#define LPP_DATATYPE_ILLUMINANCE 0x65

Adafruit_SHT31 sht31;
Adafruit_TSL2591 tsl = Adafruit_TSL2591(2591);

void setup( void )
{

    Serial.begin(115200);
    sht31.begin(0x44);
    
    tsl.begin();
    tsl.setGain(TSL2591_GAIN_MED);
    tsl.setTiming(TSL2591_INTEGRATIONTIME_100MS);
    
  
    LoRaWAN.begin(EU868);
    LoRaWAN.addChannel(3, 867100000, 0, 5);
    LoRaWAN.addChannel(4, 867300000, 0, 5);
    LoRaWAN.addChannel(5, 867500000, 0, 5);
    LoRaWAN.addChannel(6, 867700000, 0, 5);
    LoRaWAN.addChannel(7, 867900000, 0, 5);
    LoRaWAN.setDataRate(5);
    LoRaWAN.joinOTAA(appEui, appKey, devEui);
}

void loop( void )
{
    
    uint16_t temp = (uint16_t) (sht31.readTemperature() * 10);
    uint16_t rh = (uint16_t) (sht31.readHumidity() * 2);
    Serial.println("Temp: "+String(temp/10));
    Serial.println("RH:   "+String(rh/2));

    uint32_t lum = tsl.getFullLuminosity();
    uint16_t ir, full;
    ir = lum >> 16;
    full = lum & 0xFFFF;
    uint16_t lux = tsl.calculateLux(full, ir);
    Serial.println("Lux:   "+String(lux));

    if (!LoRaWAN.busy() && LoRaWAN.joined())
    {
        int lpp_channel = 0;
        LoRaWAN.beginPacket();
        LoRaWAN.write(lpp_channel++);
        LoRaWAN.write(LPP_DATATYPE_TEMPERATURE);
        LoRaWAN.write((temp >> 8) & 0xFF);
        LoRaWAN.write(temp & 0xFF);
        LoRaWAN.write(lpp_channel++);
        LoRaWAN.write(LPP_DATATYPE_HUMIDITY);
        LoRaWAN.write(rh & 0xFF);
        LoRaWAN.write(lpp_channel++);
        LoRaWAN.write(LPP_DATATYPE_ILLUMINANCE);
        LoRaWAN.write((lux >> 8) & 0xFF);
        LoRaWAN.write(lux & 0xFF);
        LoRaWAN.endPacket();
    }

    STM32L0.stop(5000);
}

kriswiner, 2018-06-22:
From Thiomas:

Can you let her/him run this WITHOUT Adafruit_TSL2591 ?

Also ask her/him, whether this is TOT off github, or installed via the json
file ?

Can you answer these questions?

kriswiner, 2018-06-23:
> Hi Thiomas,
>
> Thank you for your reply.
>
> Test it with a simple sketch and it works sleep() either stop() methods:
> `#include <STM32L0.h>
>
> void setup() {
> // put your setup code here, to run once:
> Serial.begin(115200);
>
> }
> void loop() {
> // put your main code here, to run repeatedly:
> Serial.println("Wake!");
> STM32L0.sleep(30000);
> }`
>
> Cannot do it works with the library LoRaWAN. Maybe due to OTAA join? I
> don't know if it has a lot of sense.
>
> I think i installed the library directly by cloning the github repository.
> I will try to update it.
>
> Thank you for your help.
>
> Regards,
> Jaime.

Try installing the latest STM32L0 core and test the simple examples to
verify function. It works for us with LoRaWAN.

jaimelaborda, 2018-06-25:
Yep, It works! Was working with an older version from May.  I updated and works!.

Thank you very much for your help! I close the issue.

Regards,

closed 2018-06-25 by jaimelaborda