kriswiner/CMWX1ZZABZ #7: Cricket Asset Tracker Compilation Error (Arduino 1.8.5)

etsarnas, 2018-07-23:
Hello:

AssetTracker_Cricket.v02 doesn't compile with the errors

AssetTracker_Cricket.v02:717: error: 'class GNSSClass' has no member named 'wakeup'
'class GNSSClass' has no member named 'sleep'

Best Regards
Manny

Here is the verbose output:
Arduino: 1.8.5 (Windows 10), TD: 1.42, Board: "Cricket-L082CZ, Serial, None, 32 MHz, Smallest Code"

C:\Arduino1.8.5\arduino-builder -dump-prefs -logger=machine -hardware C:\Arduino1.8.5\hardware -hardware C:\Users\etsar\AppData\Local\Arduino15\packages -tools C:\Arduino1.8.5\tools-builder -tools C:\Arduino1.8.5\hardware\tools\avr -tools C:\Users\etsar\AppData\Local\Arduino15\packages -built-in-libraries C:\Arduino1.8.5\libraries -libraries C:\Users\etsar\Documents\Arduino\libraries -fqbn=TleraCorp:stm32l0:Cricket-L082CZ:usb=cdc,dosfs=none,speed=32,opt=os -vid-pid=0X1209_0X6664 -ide-version=10805 -build-path C:\Users\etsar\AppData\Local\Temp\arduino_build_598570 -warnings=none -build-cache C:\Users\etsar\AppData\Local\Temp\arduino_cache_537538 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\etsar\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose C:\Arduino1.8.5\libraries\CMWX1ZZABZ\CricketAssetTracker\AssetTracker_Cricket.v02\AssetTracker_Cricket.v02.ino
C:\Arduino1.8.5\arduino-builder -compile -logger=machine -hardware C:\Arduino1.8.5\hardware -hardware C:\Users\etsar\AppData\Local\Arduino15\packages -tools C:\Arduino1.8.5\tools-builder -tools C:\Arduino1.8.5\hardware\tools\avr -tools C:\Users\etsar\AppData\Local\Arduino15\packages -built-in-libraries C:\Arduino1.8.5\libraries -libraries C:\Users\etsar\Documents\Arduino\libraries -fqbn=TleraCorp:stm32l0:Cricket-L082CZ:usb=cdc,dosfs=none,speed=32,opt=os -vid-pid=0X1209_0X6664 -ide-version=10805 -build-path C:\Users\etsar\AppData\Local\Temp\arduino_build_598570 -warnings=none -build-cache C:\Users\etsar\AppData\Local\Temp\arduino_cache_537538 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\etsar\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose C:\Arduino1.8.5\libraries\CMWX1ZZABZ\CricketAssetTracker\AssetTracker_Cricket.v02\AssetTracker_Cricket.v02.ino
Using board 'Cricket-L082CZ' from platform in folder: C:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7
Using core 'arduino' from platform in folder: C:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7
Detecting libraries used...
"C:\Users\etsar\AppData\Local\Arduino15\packages\adafruit\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib -fno-rtti -fno-exceptions  -w -x c++ -E -CC -D_SYSTEM_CORE_CLOCK_=32000000L -DARDUINO=10805 -DARDUINO_STM32L0_CRICKET -DARDUINO_ARCH_STM32L0  -DSTM32L082xx -march=armv6-m -mthumb -mabi=aapcs -mfloat-abi=soft -fsingle-precision-constant -DUSB_VID=0x1209 -DUSB_PID=0x6664 -DUSB_DID=0xffff '-DUSB_MANUFACTURER="Tlera Corporation"' '-DUSB_PRODUCT="Cricket"' '-DUSB_TYPE=USB_TYPE_CDC' -DDOSFS_SDCARD=0 -DDOSFS_SFLASH=0 "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7/system/CMSIS/Include" "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7/system/CMSIS/Device/ST/STM32L0xx/Include" "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7/system/STM32L0xx/Include" "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\cores\arduino" "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\variants\Cricket-L082CZ" "C:\Users\etsar\AppData\Local\Temp\arduino_build_598570\sketch\AssetTracker_Cricket.v02.ino.cpp" -o "nul"
"C:\Users\etsar\AppData\Local\Arduino15\packages\adafruit\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib -fno-rtti -fno-exceptions  -w -x c++ -E -CC -D_SYSTEM_CORE_CLOCK_=32000000L -DARDUINO=10805 -DARDUINO_STM32L0_CRICKET -DARDUINO_ARCH_STM32L0  -DSTM32L082xx -march=armv6-m -mthumb -mabi=aapcs -mfloat-abi=soft -fsingle-precision-constant -DUSB_VID=0x1209 -DUSB_PID=0x6664 -DUSB_DID=0xffff '-DUSB_MANUFACTURER="Tlera Corporation"' '-DUSB_PRODUCT="Cricket"' '-DUSB_TYPE=USB_TYPE_CDC' -DDOSFS_SDCARD=0 -DDOSFS_SFLASH=0 "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7/system/CMSIS/Include" "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7/system/CMSIS/Device/ST/STM32L0xx/Include" "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7/system/STM32L0xx/Include" "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\cores\arduino" "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\variants\Cricket-L082CZ" "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\libraries\STM32L0\src" "C:\Users\etsar\AppData\Local\Temp\arduino_build_598570\sketch\AssetTracker_Cricket.v02.ino.cpp" -o "nul"
"C:\Users\etsar\AppData\Local\Arduino15\packages\adafruit\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib -fno-rtti -fno-exceptions  -w -x c++ -E -CC -D_SYSTEM_CORE_CLOCK_=32000000L -DARDUINO=10805 -DARDUINO_STM32L0_CRICKET -DARDUINO_ARCH_STM32L0  -DSTM32L082xx -march=armv6-m -mthumb -mabi=aapcs -mfloat-abi=soft -fsingle-precision-constant -DUSB_VID=0x1209 -DUSB_PID=0x6664 -DUSB_DID=0xffff '-DUSB_MANUFACTURER="Tlera Corporation"' '-DUSB_PRODUCT="Cricket"' '-DUSB_TYPE=USB_TYPE_CDC' -DDOSFS_SDCARD=0 -DDOSFS_SFLASH=0 "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7/system/CMSIS/Include" "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7/system/CMSIS/Device/ST/STM32L0xx/Include" "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7/system/STM32L0xx/Include" "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\cores\arduino" "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\variants\Cricket-L082CZ" "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\libraries\STM32L0\src" "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\libraries\LoRaWAN\src" "C:\Users\etsar\AppData\Local\Temp\arduino_build_598570\sketch\AssetTracker_Cricket.v02.ino.cpp" -o "nul"
"C:\Users\etsar\AppData\Local\Arduino15\packages\adafruit\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib -fno-rtti -fno-exceptions  -w -x c++ -E -CC -D_SYSTEM_CORE_CLOCK_=32000000L -DARDUINO=10805 -DARDUINO_STM32L0_CRICKET -DARDUINO_ARCH_STM32L0  -DSTM32L082xx -march=armv6-m -mthumb -mabi=aapcs -mfloat-abi=soft -fsingle-precision-constant -DUSB_VID=0x1209 -DUSB_PID=0x6664 -DUSB_DID=0xffff '-DUSB_MANUFACTURER="Tlera Corporation"' '-DUSB_PRODUCT="Cricket"' '-DUSB_TYPE=USB_TYPE_CDC' -DDOSFS_SDCARD=0 -DDOSFS_SFLASH=0 "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7/system/CMSIS/Include" "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7/system/CMSIS/Device/ST/STM32L0xx/Include" "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7/system/STM32L0xx/Include" "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\cores\arduino" "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\variants\Cricket-L082CZ" "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\libraries\STM32L0\src" "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\libraries\LoRaWAN\src" "-IC:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\libraries\TimerMillis\src" "C:\Users\etsar\AppData\Local\Temp\arduino_build_598570\sketch\AssetTracker_Cricket.v02.ino.cpp" -o "nul"
[... cut ...]
Using library RTC at version 1.0 in folder: C:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\libraries\RTC 
Using library Wire at version 1.0 in folder: C:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\libraries\Wire 
Using library SPI at version 1.0 in folder: C:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\libraries\SPI 
Using library CayenneLPP at version 1.0.0 in folder: C:\Users\etsar\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\libraries\CayenneLPP 
exit status 1
'class GNSSClass' has no member named 'sleep'

kriswiner, 2018-07-23:
Yes, we revised the API and now use GNSS.suspend and GNSS.resume. You can
see this in the latest Cricket sketch here
<https://github.com/kriswiner/CMWX1ZZABZ/blob/master/CricketAssetTracker/AssetTracker_Cricket/Examples/AssetTracker_Cricket.v04.ino>

etsarnas, 2018-07-25:
Works now, thanks.

Regards
Manny

closed 2018-07-26 by etsarnas