kriswiner/CMWX1ZZABZ #10: advice on ttn OTAA integration?

everhamme, 2019-01-14:
I compiled the sketch just fine for the LoRa tile project (after updating SPIFlash.h to SPIFlashClass.h), but having issues with TTN.  I've got the right keys entered in the sketch and the node joins OK (as shown in the device Application data log), but no payload data is delivered to my application.  I can see the data payload coming in to my gateway on TTN, but not making its way to the application.  btw.. The Framecounter is not incrementing above 0.  I can ask on TTN if that's more appropriate!

kriswiner, 2019-01-14:
Firstly, I would recommend you use this
<https://github.com/kriswiner/CMWX1ZZABZ/tree/master/Cicada> sketch, it is
the latest for this board.

Secondly, for TTN this line:

LoRaWAN.setSubBand(1); // 1 for MTCAP and 2 for TTN gateway

needs to be set to 2. TTN network for some reason expects subBand 2.

everhamme, 2019-01-14:
Thanks....  I just tried using the Cicada sketch instead of the LoRa sensor tile v.05 sketch..  I did have to change the subband to 2.  I get the same result in TTN.  My node shows up as having joined... as verified by the traffic in the gateway log console and application/device log console... but no data is transmitted to the application/device... as verified by seeing payload data coming in to the ttn gateway console (my gateway 50 ft away).. and no payload data coming in to the application/device log console...  i'll ask on ttn why data is slow to show up in my application

patmolloy, 2019-01-14:
Also worth disabling “Frame Counter Checks” .. this is frowned up as a security risk, but in my experience is sometimes the only way!

Pat ~Mobile~ iPhone XS Max

kriswiner, 2019-01-14:
I wouldn't expect switching to the Cicada sketch to ficx this problem since
I suspect it is on the TTN side, it's just the latest sketch for this board
so some of the low power features are done better.

You do need to sue subBand 2 for the TTN netwrk to recognize the data
packets or some such. Maybe powering everything down and restarting would
help, don't know.

If the node is joined and you see data coming to the gateway this is
clearly a forwarding issue not an end node issue. Sorry I don't know how to
fix this.

patmolloy, 2019-01-14:
Try disabling  "Frame Counter Checks" 
[image]

everhamme, 2019-01-14:
I tried disabling them..  no luck...  I even tried the example OTAA sketch in the LoraWAN folder library....  OTAA is not easy to troubleshoot on TTN...

kriswiner, 2019-01-14:
Just checking, but you do have an antenna mounted on the Cicada, right?

kriswiner, 2019-01-14:
> Kinda low, I usually see ~ -40 - 50 when within a meter of the MTCAP
> gateway. But this isn't the problem if the gatway is receiving
> packets...odd.

Are you using the CayenneLPP syntax for the packet bytes? TTN won't
recognize these I believe.

everhamme, 2019-01-14:
yes... the example sketch you have uses Cayenne... TTN should still shown the data traffic to the application i believe...

kriswiner, 2019-01-14:
LoRaSensorTile.v05 is set up to use TTN and doesn't use Cayenne, so I think
you are right. There is something wrong on the TTN side....

everhamme, 2019-01-17:
For reference the fix was having my application and gateway set to the same server.