kriswiner/CMWX1ZZABZ #11: GNAT STM32 bootloader/usb driver problem

everhamme, 2019-01-16:
I connected my new gnat to my computer just fine and compiled a new example sketch from the github.  I was able to select the Cricket board from the board manager, correct com port, and load the new sketch just fine.  After the sketch loaded I went to open the serial monitor and arduino ide said the com port wasn't found.  If I look in my device manager (win 10 64 bit) I only see the STM32 BOOTLOADER showing up under USB devices.  No entry showing up in the Ports section.  I did the reset option (hold down boot and toggle reset).  This doesn't fix the issue.  I can hook up my cicada/loratile device and it shows up just fine.  any other troubleshooting things to try to get the com port back?

kriswiner, 2019-01-16:
Download the latest Arduino core and use the Gnat variant. The Cricket
variant will not work.

everhamme, 2019-01-16:
Oops..  I should have selected gnat..  how do I get com port back?

kriswiner, 2019-01-16:
Close Arduino IDE. Press and hold BTN, press and release RST, release BTN.
Open Arduino IDE, select Gnat board variant, flash a blink program (no COM
port should show). After flash the laptop will assign a COM port.

kriswiner, 2019-01-16:
> Working!
> Author <author@noreply.github.com<mailto:author@noreply.github.com>>
> <mailto:notifications@github.com%0b>> <mailto:notifications@github.com>>

Yup, if you ever load a program with the wrong board variant and get stuck
this procedure will work.