kriswiner/CMWX1ZZABZ #12: txPower and DataRate settings

everhamme, 2019-01-17:
I'd like to modify seteDataRate and setTxPower.   Could you document valid range of values in  // Configure LoRaWAN connection?  I tried hunting down the forks and its not clear..  Here's what I found 

Valid DataRate
 *    LoRaWAN US or AU:
 *    
 *    0: SF = 10, BW = 125 kHz, BitRate =   980 bps
 *    1: SF =  9, BW = 125 kHz, BitRate =  1760 bps
 *    2: SF =  8, BW = 125 kHz, BitRate =  3125 bps
 *    3: SF =  7, BW = 125 kHz, BitRate =  5470 bps

Valid TxPower

I believe its expressed in dB
0 to 20 (eu) ...  maybe 0 to 14 (us) 

Let me know!

kriswiner, 2019-01-17:
The way I understand it is that in the LoRaWAN protocol, these are set by
the gateway in downlink messages to the end node. But from the board
variant source files:

  int LoRaWANClass::setTxPower(float power)
{
    unsigned int txPower;
    MibRequestConfirm_t mibReq;

    if (!_Band) {
        return 0;
    }

    if (_tx_busy) {
        return 0;
    }

    if (_Band->Region == LORAWAN_REGION_US915) {
        txPower = (floorf(30.0f - power) + 1) / 2;
    } else {
        txPower = (floorf(LoRaMacParams.MaxEirp - power) + 1) / 2;
    }
etc.

    int setTxPower(float power); // 2dm to 20dbm

data rate looks right.

Is there some documenttion of this Thomas?

kriswiner, 2019-01-17:
A bit of googling and I found this:

https://blog.dbrgn.ch/2017/6/23/lorawan-data-rates/