kriswiner/CMWX1ZZABZ #14: Issue with PA12

PAk-CatchFire, 2019-02-15:
Hi there. I am having an issue using a B-L072Z-LRWAN1 board and wanted to use PA12 as an output pin. I have checked variant cpp and added (in the digital block - position 3) the line:

` { GPIOA, STM32L0_GPIO_PIN_MASK(STM32L0_GPIO_PIN_PA12), STM32L0_GPIO_PIN_PA12, (PIN_ATTR_EXTI), PWM_INSTANCE_NONE, PWM_CHANNEL_NONE, ADC_CHANNEL_NONE },`

I have alse changed:

```
void initVariant()
{
    //CMWX1ZZABZ_Initialize(STM32L0_GPIO_PIN_PA12, STM32L0_GPIO_PIN_PA11);
CMWX1ZZABZ_Initialize(STM32L0_GPIO_PIN_NONE, STM32L0_GPIO_PIN_PA11);

}
```
Then in variant.h I have added:
`#define PIN_D3 (03ul)`
but with no luck.

On the other hand, I have done the same for PA5 (as Analog1 - A1) and it worked fine for that pin.

Is there anything I am missing?
Thank you

kriswiner, 2019-02-15:
No idea. Isn't PA12 used for USB (D+)?

Are you using the STM32L0 core here
<https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0>?

PAk-CatchFire, 2019-02-15:
Yes, I am using that code. I don't know if USB is automatically enabled....but I think it shouldn't.
I cannot see why it isn't working. Did you need to change the variant files?
Regards