kriswiner/CMWX1ZZABZ #16: Sigfox

jpmeijers, 2019-03-18:
The Murata modules has Sigfox support. Is it at all possible to do Sigfox on the Cricket?

kriswiner, 2019-03-18:
I don't think so, but Gnat supports SigFox.

kriswiner, 2019-03-18:
SFOX is only possible on GNAT.

Highlanderes, 2019-09-04:
The Murata data sheet states in 7.3, that their SigFox FW requires pin 1 connected to pin 48, and pin 10 connected to pin 21, and pin 47 connected to pin 46.

In the schematics on Tindie that is not the case (at least GNAT, Cricket and Grasshopper).

The sheet also states in 9 that for LoRa, pin 48 (VDD_TCXO), should be connected to VDD or pin 1 (PA16), which it is not on the Tindie schematics, and LoRa works anyway, so perhaps the Data sheet is just not precise or is referring to another SW?

kriswiner, 2019-09-04:
Lots of ways to manage the TCXO. Sigfox can work on Gnat, but not the
others.

Highlanderes, 2019-09-06:
I see, VDD_TCXO is defined in the Arduino core variant. Why only GNAT then and not eg. Grasshopper?

kriswiner, 2019-09-06:
Because we did Gnat later and differently.

Highlanderes, 2019-09-07:
Has it been verified?

I prefer the more generel purpose Grasshopper for my new sensor project, and was hoping to learn if there was an easy mod to also use the Grasshopper for SigFox later.

kriswiner, 2019-09-07:
Verified? Not sure what you refer to or what verified means but SigFox can
work on the Gnat.

If you want to use SigFox with something like a Grasshopper, you can modify
the design (open source) to meet SigFox configuration requirements (see
Murata data sheet).