kriswiner/CMWX1ZZABZ #17: GNSS always busy--can't progress

TFenby, 2019-06-05:
I'm trying your simple GNSS test/example, and finding that the first loop of `while (GNSS.busy()) { }` never exits. I checked the source and found that it returns the value of gnss_busy(), but can't find any source for that--I assume it's part of the GPS chip? Regardless, do you know what might be causing it never not to be "busy"?

kriswiner, 2019-06-05:
Which device? And specifically which sketch (link please)?

TFenby, 2019-06-05:
Long Cricket. This sketch and line in particular: https://github.com/kriswiner/CMWX1ZZABZ/blob/c4948da7bfa3430b15b012d4e20dac2a75118cf3/longCricketAssetTracker/AssetTracker_longCricket.v04i/AssetTracker_longCricket.v04i.ino#L183

TFenby, 2019-06-05:
There's also the GPS example that came with the Arduino IDE package via https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0

kriswiner, 2019-06-05:
Are you using the Arduino IDE? Did you select the Cricket board variant to
program it?

TFenby, 2019-06-05:
Yes to both questions. I've actually successfully registered the LoRaWAN portion with TTN at this point. Just can't figure out what's up with GNSS.

kriswiner, 2019-06-05:
You could try setting pinMode(GNSS_en, OUTPUT); and digitalWrite(GNSS_en,
HIGH); but this should be automatic. Are you suing the latest Tlera Corp
boards version? Maybe update to the latest version?

TFenby, 2019-06-05:
Is there another repository for that besides https://github.com/GrumpyOldPizza? I've looked through your GitHub, Tindie, and even HAD but have only really found your sketches and GrumpyOldPizza's Arudino SDK board info.

kriswiner, 2019-06-05:
Hnh? How are you programming this board then?

Use the Arduino IDE, load the Tlera Corp boards using the board manager and
select the Cricket board variant for this sketch to work.

TFenby, 2019-06-05:
Yeah, that's what I've done. Tlera Corp didn't show up in the boards manager, though, until I added https://grumpyoldpizza.github.io/ArduinoCore-stm32l0/package_stm32l0_boards_index.json as a board manager URL in the IDE. Perhaps that's where I went astray?

kriswiner, 2019-06-05:
No that's right. Did you select the Cricket variant? Are you using v.0.0.10
(the latest)?

TFenby, 2019-06-05:
Yes to both. Sounds like I haven't made any obvious mistakes so far, so that's heartening at least. I think for the time being I'm going to focus back in on TTN and come back to this with a fresher mind. If I continue to have issues I'll reach out again. Thank you for your very rapid responses!

kriswiner, 2019-06-05:
Do you have an active patch antenna on the longCricket?

Try setting pin 5 to HIGH, this turns on the LDO powering the GNSS engine.
It is possible yours is defective but I ran this program on your board
before shipping to you so I know it worked at least once.

TFenby, 2019-06-05:
I'll give pin 5 a shot. And I do have an external antenna on order now to see how that affects it. Will let you know the results.

kriswiner, 2019-06-06:
Select internal antenna if not using the external one in set up.