kriswiner/CMWX1ZZABZ #18: LongCricket

patmolloy, 2019-06-16:
This is up and running successfully, but I am getting possibly wrong data from the BME280. I have the device transmitting every minute for now. The values a) look wrong and b) don't seem to change much/at all over time. GPS looks good. The analog port is returning a value close to 0.00 .. is that right or should it be the battery voltage? I've attached a data dump for the last hr from cayenne ..

The device is in free air, not enclosed.

At the moment external temp is circa 13.5 degC, humidity at 92% (rainy).

Overnight the device was reporting 20 degC all night - i.e. incorrect and not changing.

Any pointers please? Thanks!

[9915c234-d25d-4801-8af3-39225d9adfbc.xlsx](https://github.com/kriswiner/CMWX1ZZABZ/files/3293982/9915c234-d25d-4801-8af3-39225d9adfbc.xlsx)

kriswiner, 2019-06-16:
Not sure what's happening here.

Looks like the temperature, pressure, and humidity are changing, or at
least do change in the output you sent me. Are you running this
<https://github.com/kriswiner/CMWX1ZZABZ/blob/master/longCricketAssetTracker/AssetTracker_longCricket.v04i/AssetTracker_longCricket.v04i.ino>
sketch?

Maybe you are asking for sensor data over a long duty cycle?

As far as the battery voltage, you would need to enable the fet on pin 2 to
allow the voltage measurement circuit to function. Are you doing this?

patmolloy, 2019-06-16:
Hi Kris,

I am just using your github sketch -- the one you referenced. Just changed the TTN stuff, and the freq to EU868 Nothing else at all. I'll reload it all again to make sure I did not screw anything up. Was not aware of the FET on pin 2 requirement but looking at the sketch I do see

`#define VbatCtl 2    // enable for battery voltage monitor`

So that looks right?

Pat

kriswiner, 2019-06-16:
Yes, the sketch should work out of the box. You might need to update the
times you want to read the sensors and send the data etc. If you run it on
the USB first you can verify that the BME280 is working, which it should
be, or at least was just before I sent it to you...

kriswiner, 2019-06-16:
> Kris, I re-downloaded, made the change for TTN and freq and re-compiled ..
> same as before as far as I can tell ..
>
> If I am on USB Power, VBAT (as transmitted to TTN) should be approx zero?
>
> Even though out of the box the sketch is transmitting every 1 minute, is
> the sampling of the sensors on some different timing? I've not looked in
> detail at the sketch yet (and most likely my coding skills are not up to it
> anyway!!)
>
> Thanks for the support, as ever :)
>
> This is the serial monitor output at startup ..
>
> Serial enabled!ID bytes: C2 28 17
> Winbond  W25Q80BLUX1G   Chip ID = 0xEF, 0x40, 0x14, 0x0
> Macronix MX25L12835FZNI Chip ID = 0xC2, 0x20, 0x18, 0xC2
> Macronix MX25R6435FZAI  Chip ID = 0xC2, 0x28, 0x17, 0xC2
> Spansion S25FL127S      Chip ID = 0x01, 0x20, 0x18, 0x4D
>
> VDDA = 3.30 V
> VBAT = 0.17 V
> USB Connected!
> STM32L0 MCU Temperature = 26.23
>
> BMA400 accelerometer...
> BMA400 I AM 90 I should be 90
>
> BME280 I AM 60 I should be 60
>
> BMA400 + BME280 are online...
>
> x-axis self test = 2412.1mg, should be > 2000 mg
> y-axis self test = 2085.9mg, should be > 1800 mg
> z-axis self test = 859.4mg, should be > 800 mg
> hold flat and motionless for bias calibration
> x-axis offset = -134.7 mg
> y-axis offset = 100.2 mg
> z-axis offset = 47.3 mg
> Calibration coefficients:
> _dig_T1 =28083
> _dig_T2 =26675
> _dig_T3 =50
> _dig_P1 =39085
> _dig_P2 =-10673
> _dig_P3 =3024
> _dig_P4 =9073
> _dig_P5 =-223
> _dig_P6 =-7
> _dig_P7 =12300
> _dig_P8 =-12000
> _dig_P9 =5000
> _dig_H1 =0
> _dig_H2 =373
> _dig_H3 =0
> _dig_H4 =292
> _dig_H5 =50
> _dig_H6 =30
> LOCATION: NONE
> SATELLITES: 2
>       G3: SNR=19, ELEVATION=0, AZIMUTH=0, ACQUIRED
>       G4: SNR=20, ELEVATION=0, AZIMUTH=0, ACQUIRED
> VBAT = 0.04 V
> BME280:
> Altimeter temperature = 24.99 C
> Altimeter temperature = 76.98 F
> Altimeter pressure = 1014.66 mbar
> Altitude = -38.56 feet
> Altimeter humidity = 45.0 %RH

The BME280 seem to be working fine. Do you have a battery installed?
Otherwise, what is the problem?

patmolloy, 2019-06-16:
No, no battery at present. Just on USB. The issue I was/am seeing with the BME280 is that overnight it reported temp and humidity values that were wrong, and which did not change. That could be because of a very slow sampling rate ..  for example right now it is showing ..

kriswiner, 2019-06-16:
The repository sketch samples the BME280 every minute.

patmolloy, 2019-06-16:
Thanks. I guess this remains a mystery then .. the values do not change when the device moves inside vs. outside and the outside value is incorrect (and unchanging). Could this be related to the device being on USB power? Or a faulty BME280? Bit stuck.

kriswiner, 2019-06-16:
I don't understand what the problem is exactly. The BME280 seems to be
working.

If you connect a 3,7 V LiPo battery you should see the battery voltage
measured by VBAT. If you then disconnect the USB cable you should see all
of the data you request sent over the LoRaWAN. As far as I can tell, based
on the data snippet you sent, this seems to be happening correctly. Is the
device in a box? Remember the BME280 will measure temperature of the board
mostly, and if the board is warmer than the outside air it could show a
discrepancy. But generally, in my testing and experience with these
devices, the temp, press, and humidity track the outside conditions.

I suggest you run it for a day, plot the temperature versus time and see
what it shows.

From what you have shown me so far, I cannot detect any error at all, and I
am a bit confused as to what you mean that it isn't working...

kriswiner, 2019-06-16:
Maybe I misunderstood.

Are you saying that the BME280 data seems correct and updates every minute
and slightly changes when indoors when the device is powered by USB but ....

once you power it with a battery and take the device outside, the BME280
temperature, pressure, and humidity values are similar to those measured
indoors, stay exactly the same and never change even one bit for hours at a
time?

In this case, it is possible the BME280 is malfunctioning but I have never
seen this behavior and really cannot understand it.

If you are not happy with the device you may return it and I will refund
your money.

patmolloy, 2019-06-16:
Hi Kris, I am just running off USB power at the moment. No battery tried so far.

I am definitely getting data back to TTN and Cayenne every minute. Including data from the BME280. But the BME data is just wrong. It's outside right now and I know what values I should be getting, but am not getting them. Moreover, over time they just do not change. Even if I bring the device inside, the values do not change. 12 hours last night it reported 20C and did not change. I got the value 20 every minute for 12 hours straight.

patmolloy, 2019-06-16:
Last 3 packets in the TTN console .. GPS changed a tiny bit. BME280 values rock solid but Temp and Humid wrong .. It's 18.2C and 73% right now (two different sources agree). 

[image]

kriswiner, 2019-06-16:
So do the values change on the laptop monitor? Is there something wrong
with TTN or more likely the changes you made to the sketch? Looks like you
are sending the initial values via TTN and probably not updating. Send me
your main sketch and I will show you the problem.

patmolloy, 2019-06-16:
Chris, here is the .ino (just saved as .txt). It is literally your example with only the following lines altered ..

```
const char *appEui = "70B3D57ED001DDB7";
const char *appKey = "ACF8ADB08F84ED97777A8973B5385173";
const char *devEui = "567890ABCDEF0102";

LoRaWAN.begin(EU868);
LoRaWAN.setADR(true);
LoRaWAN.setSubBand(2); // 1 for MTCAP, 2 for TT gateways

```
The serial debug monitor does not output data for the BME280 except at startup ...

[AssetTracker_longCricket.txt](https://github.com/kriswiner/CMWX1ZZABZ/files/3294401/AssetTracker_longCricket.txt)

patmolloy, 2019-06-16:
Here is the SerialMonitor output on boot ...
[SerialMonitor.txt](https://github.com/kriswiner/CMWX1ZZABZ/files/3294415/SerialMonitor.txt)

kriswiner, 2019-06-16:
"The serial debug monitor does not output data for the BME280 except at
startup ..."

It should output this data every minute. Are you saying it never does after
the initial report?

patmolloy, 2019-06-16:
Don't see it .. there is a *lot* of GPS chatter, but I cannot see anything else after the initial report.

kriswiner, 2019-06-16:
Very odd, try one second interval:

change this:

  RTC.enableAlarm(RTC.MATCH_SS);  // alarm once per minute
//    RTC.enableAlarm(RTC.MATCH_ANY); // alarm once a second

to this:

  //RTC.enableAlarm(RTC.MATCH_SS);  // alarm once per minute
    RTC.enableAlarm(RTC.MATCH_ANY); // alarm once a second

You should see BME280 data every second

patmolloy, 2019-06-16:
Ok, will do. Searching thru I did actually find one instance .. buried in a lot of GPS logging ..

```
LOCATION: 3D 2019/6/16 18:09:05.000 17D LLA=51.4572334,-0.2193177,34.605 EPE=6.879,6.225 SATELLITES=6 DOP=2.80,2.80

SATELLITES: 12
      G1: SNR=25, ELEVATION=83, AZIMUTH=325, EPHEMERIS, ACQUIRED, LOCKED, NAVIGATING
      G3: SNR=42, ELEVATION=50, AZIMUTH=227, ALMANAC, EPHEMERIS, ACQUIRED, LOCKED, NAVIGATING
      G4: SNR=0, ELEVATION=0, AZIMUTH=0
      G8: SNR=37, ELEVATION=30, AZIMUTH=165, EPHEMERIS, ACQUIRED, LOCKED, NAVIGATING
     G11: SNR=33, ELEVATION=78, AZIMUTH=162, EPHEMERIS, ACQUIRED, LOCKED, NAVIGATING
     G22: SNR=35, ELEVATION=73, AZIMUTH=215, EPHEMERIS, ACQUIRED, LOCKED, NAVIGATING
     G28: SNR=33, ELEVATION=28, AZIMUTH=275, EPHEMERIS, ACQUIRED, LOCKED, NAVIGATING
     G32: SNR=23, ELEVATION=0, AZIMUTH=0, ACQUIRED
    S120: SNR=0, ELEVATION=29, AZIMUTH=199
    S123: SNR=37, ELEVATION=24, AZIMUTH=142, ACQUIRED, LOCKED
    S136: SNR=0, ELEVATION=31, AZIMUTH=173
    R???: SNR=35, ELEVATION=0, AZIMUTH=0, ACQUIRED
VBAT = 0.03 V
BME280:
Altimeter temperature = 19.62 C
Altimeter temperature = 67.32 F
Altimeter pressure = 1015.10 mbar
Altitude = -50.46 feet
Altimeter humidity = 51.9 %RH
 
RTC:
RTC Time = 18:09:05.627 
RTC Date = 19:6:47

LOCATION: 3D 2019/6/16 18:09:06.000 17D LLA=51.4572334,-0.2193200,34.455 EPE=6.816,6.197 SATELLITES=6 DOP=2.80,2.80
```

kriswiner, 2019-06-16:
OK, seems like it is working. Then why is this not showing up on TTN?

patmolloy, 2019-06-16:
Kris, here is what happened after that change .. The BME280 values look a lot more sensible (and are wobbling about a bit, which I would expect)

Shortly after the "***GNSS go to sleep!***" toward the end of the dump, the logging stopped (after the line "R???: SNR=36, ELEVATION=0, AZIMUTH=0, ACQUIRED, LOCKED"

[SerialMonitor2.txt](https://github.com/kriswiner/CMWX1ZZABZ/files/3294457/SerialMonitor2.txt)

kriswiner, 2019-06-16:
Are you using the latest L0 core? Might load this and try again.

patmolloy, 2019-06-16:
BTW The logging did start up again, but contained only GPS stuff .. nothing else ...

Am using https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0

patmolloy, 2019-06-16:
Ardunio config ..

[image]

patmolloy, 2019-06-16:
Kris, I have to do some prep work for a project tomorrow (it's heading for 8pm here). WIll be back online later .. thanks for all the help and sorry for any trouble!

kriswiner, 2019-06-16:
In the Boards manager you want to update the Tlera Boards core to v.0.0.10
see attached.

I think you are encountering a bug long fixed in a more recent version of
the core.

kriswiner, 2019-06-16:
[image]

patmolloy, 2019-06-16:
Kris, where do I find this .. what is the URL I need to put into Additional Board Manager?

kriswiner, 2019-06-16:
Go to the board manager, select TleraCorp boards, and update to the latest.
Don't you have the JSON in the preferences?

kriswiner, 2019-06-16:
Like this...

patmolloy, 2019-06-16:
Kris, I didn't have it .. but do now.  I had manually created the folder/contents in arduino/hardware before .. 

Just re-compiling ...

patmolloy, 2019-06-16:
Voila ! It works :)