kriswiner/CMWX1ZZABZ #19: How Can I receive LoRa Data?

paengpd, 2019-06-21:
Do you have Example code?
TX is so good.
but RX is not good.

I use below code.
but can't enter.

int packetSize = LoRaWAN.parsePacket();
{
}

Any ideas?

kriswiner, 2019-06-21:
Try the LoRaWAN_Callbacks example for Grasshopper.

paengpd, 2019-06-24:
I want use only two grasshopper.
If one grasshopper did Join request, other grasshopper have to Join accept.
But I can't find Join accept example.
Do I must use geteway or network server?
Grasshopper can't do that?

kriswiner, 2019-06-24:
No, you can use LoRa Radio or FSK examples, check for ping_pong example,
then no gateway required.

paengpd, 2019-07-22:
I did it for use Radio example.
but I want receive data for use LoRaWAN.
Can you tell me good idea?

kriswiner, 2019-07-22:
For LoRaWAN you need a gateway.

closed 2019-09-10 by paengpd