kriswiner/CMWX1ZZABZ #20: Hang on LoRaWAN.joinOTAA(appEui, appKey, devEui);

jacky4566, 2019-07-19:
Hi guys,

Just getting started with this core and im having a problem with the LoRaWANtest.ino for the grasshopper.

When i put in my keys and run the example it will hang on join. I have verified this with LED blinks.

Where in the code should i start digging to diagnose this? OR what am i missing?

Note that I am not actually using a grasshopper but the ST dev board. B-L072Z-LRWAN1. I have reconfigured the board to operate in standalone mode so it should work the same as a grasshopper.

kriswiner, 2019-07-19:
" I am not actually using a grasshopper but the ST dev board.
B-L072Z-LRWAN1"

Why not use the dev board variant?

" it should work the same as a grasshopper"

Nope...

jacky4566, 2019-07-19:
Its what I was given to work with.
The B-L072Z and the grasshopper both use the CMWX1ZZABZ module and have similar wiring so they should functionally work the same. 
PA8 is tied high for USB on this core.
TCXO_OUT is tied to OSC_IN as per the grasshopper drawing.

VDD_TCXO is tied high on this board instead of OSC_OUT on the grasshopper. Could that be it?

I can get the MCU to function properly, serial, temps, etc. But comms with the SX chip just halt.

kriswiner, 2019-07-19:
If you are using the L0 Arduino core from grumpyoldpizza just select the
proper board variant for your board; it is not Grasshopper.

Otherwise it will not work.

jacky4566, 2019-07-19:
I have resolved the issue.

closed 2019-07-19 by jacky4566

kriswiner, 2019-07-19:
How?