kriswiner/CMWX1ZZABZ #22: Question: Is it possible to change the callback delay?

applecrusher, 2019-11-11:
I am working with the Grasshopper development board. I am using callbacks for uplinks using the example code in this library. This works well, but the callback delay seems to be exactly 3 seconds. I do not know if this is hardcoded somewhere, or if this is actually the time it takes to complete what is necessary to reach the gateway. I am very close to the gateway. Is it possible to shorten this callback time? I am trying to send burst messages and to shorten the time the STM32L0 and antenna are on. I do not know the best way to do so with the code and any help would be greatly appreciated! Thank you!

kriswiner, 2019-11-11:
Not sure, but remember, for each Tx there are two Rx, which is part of the
LoRaWAN protocol and this entire message scheme can take 3 seconds I
suppose.. This has nothing to do with proximity of the gateway and would be
the same if the gateway were 500 meters away. If you have a power meter you
will see that the peak current of ~30 mA only lasts a few tens of ms on Tx,
so any attempt to shorten the entire packet time won;t save any power.

applecrusher, 2019-11-11:
Great! Thank you for the good answer and quick response

closed 2019-11-11 by applecrusher