kriswiner/CMWX1ZZABZ #23: Firmware LoRaSensorTile.v05

klyaruqui, 2020-03-06:
Hello kris.

I wanna know if I can update the firmware of the LoRaSensorTile.v05 to program in python or another language. Because arduino ide consume more resources

kriswiner, 2020-03-06:
No idea, I use the L082 system layer/Arduino core to program these.

klyaruqui, 2020-03-06:
thanks Kris 

I'll ask to another question in the future :)

kriswiner, 2020-03-06:
Not sure what your concern is with the Arduino core for the L082. It is
power and memory efficient, robust, and complete. What resources do you
think are being consumed?

klyaruqui, 2020-03-09:
Thanks Kris I will do with the Arduino  core

egionet, 2020-04-03:
Hello Kris,

I installed board support for the sensor tile per instructions and downloaded the entire library (i.e. CMWX1ZZABZ) via github.  I updated the v05 sketch (LoRa settings to match TTN settings), referenced the SPIFlash library appropriately (i.e. SPIFlashClass), and the sketch compiles.  However, when I attempt to upload the sketch to the board it fails.  The programmer in the sketch defaulted to "AVRISP mkII" but is this correct?

BTW: Arduino IDE 1.8.10 and outputs are appearing via Serial Monitor

Thanks

Eric

kriswiner, 2020-04-03:
The programmer is not used so irrelevant.

Did you select Cicada as the board variant?

What is the output on programming failure?

GrumpyOldPizza, 2020-04-04:
Do you have the 32 bit library support (like libc and such) installed on Ubuntu ?

egionet, 2020-04-04:
I will check.  The board works fine on Windows after updating via Zadig and finally seeing data on TTN through the Dragino LPS8.