kriswiner/CMWX1ZZABZ #24: LORAWAN_DEFAULT_PORT override

olivierfavre, 2020-04-27:
Evening!
Thanks for the great job!
Two questions (GNAT):
1. Is it possible to override in the sketch, not the library LORAWAN_DEFAULT_PORT ?
2. The NoMotionActivityTimer and InMotionActivityTimer cycles seem to have no effect when the GNAT is powered by USB, normal?
Great device !

kriswiner, 2020-04-27:
Not sure about 1).

The Gnat cannot enter stop mode when USB is connected.

kriswiner, 2020-04-27:
Still, not sure why the noMotion functions shouldn't work. How do you know
they don't, what is the symptom?

VinceICPO, 2020-11-29:
Looking at the library, I see this is what "LoRaWAN.beginPacket(3);" does. If you add this to your sketch, default port becomes "3". Valid values are 1..223.
Quote from[ Lora dev portal](https://lora-developers.semtech.com/library/tech-papers-and-guides/the-book/the-port-field/):
The LoRaWAN specification provides a data delivery field (FRMPayload) and a Port field (FPort) to distinguish between different types of messages. FPort 0 is reserved for MAC messages. Do not use this port to transport data fields. FPort 224 is reserved for MAC compliance testing and FPorts 225-255 are reserved for future standardized application extensions1. This leaves valid port numbers between 1 and 223.