kriswiner/CMWX1ZZABZ #25: CAD detected

pipesg, 2020-05-25:
Hello,
I am using your boards (grasshopper and cricket) to develop an animal tracker by using the LoRa radio not LoRaWAN. To reduce the power consumption from the tracker I will like to use a low energy design provide by Semtech (AN1200.17) using the channel activity detection (CAD) of the radio. I am not very proficient in programming so perhaps I am missing something or doing something incorrect, but I am not able to detect the preambles. I have also asked this in  GrumpyOldPizza/ArduinoCore-stm32l0 repository but without replies, so sorry for duplicate the question.
I am using two simple programs, one to send a message with a long preamble (10 seconds) and another implementing the CAD function;
Preamble send
`#include "LoRaRadio.h"
#define myLed 13

void setup( void )
{
    Serial.begin(115200);
    pinMode(myLed, OUTPUT);
    digitalWrite(myLed, LOW);
    LoRaRadio.begin(865200000);
    LoRaRadio.setFrequency(865200000);
    LoRaRadio.setTxPower(14);
    LoRaRadio.setPreambleLength (300);
    LoRaRadio.setBandwidth(LoRaRadio.BW_125);
    LoRaRadio.setSpreadingFactor(LoRaRadio.SF_12);
    LoRaRadio.setCodingRate(LoRaRadio.CR_4_5);
    LoRaRadio.setLnaBoost(true);
    delay (2000);
}

void loop( void )
{
   unsigned long previous = millis ();
   digitalWrite(myLed, HIGH);
   LoRaRadio.beginPacket();
   LoRaRadio.write(1);
   LoRaRadio.endPacket();
   while (LoRaRadio.busy());
   unsigned long last = millis ();
   last -= previous;
   Serial.println (last);
   digitalWrite(myLed, LOW);
   delay(3000);
}`

CAD
`#include "LoRaRadio.h"
bool CadDone;
bool CadDetected;

void setup( void ){
Serial.begin(115200);
LoRaRadio.begin(865200000);
LoRaRadio.setFrequency(865200000);
LoRaRadio.setTxPower(14);
LoRaRadio.setBandwidth(LoRaRadio.BW_125);
LoRaRadio.setSpreadingFactor(LoRaRadio.SF_12);
LoRaRadio.setCodingRate(LoRaRadio.CR_4_5);
LoRaRadio.setLnaBoost(true);
LoRaRadio.onCad(CADcallback);
}

void loop( void ){
CadDone = false;
CadDetected = false;
Serial.print ("Caddone: "); Serial.println (CadDone);
unsigned long cadinit = millis ();
LoRaRadio.cad ();
while (LoRaRadio.busy());
unsigned long cadend = millis ();
Serial.print ("Caddone: "); Serial.println (CadDone);
Serial.print ("CadDetected: "); Serial.println (CadDetected);
cadend -= cadinit;
Serial.print ("Cad length: "); Serial.println (cadend);
Serial.println ();
delay (1500);
}

void CADcallback (void){
CadDone = true;
CadDetected = LoRaRadio.cadDetected ();
}`

I am quite that the CAD is executing because the the caend-cadinit lenght is the same as the estimated time in the LoRa calculator by Semtech but the program is not able to detect the preambles by using LoRaRadio.cadDetected () function.
Any help would be greatly welcome, thanks in advance!

kriswiner, 2020-05-25:
Sorry, I have never used the CAD function before and I am not willing to
try and debug your code.

Are you using a Grasshopper?

Why are you using PABoost? Is this supported by your hardware?

Can you get plain old LoRaRadio to work?

On first read the logic seems off in your CAD example but I might just be
missing something...

pipesg, 2020-05-26:
Hi,
Many thanks for your prompt reply.
I am using two grasshoppers two test this example and LoRaRadio seems to work without problems in sending and receiving. I have also tested the sense function and it also works, only cadDetected looks like fail or I am using it incorrectly.  Cad and caddone apparently works fine.
Why you say old LoRaRadio, is there another library? Why you think the logic is off?
Regarding the PAboost I have just copied the setup part from one of the examples of the LoRaRadio to test cad, in all of them are enabled.
One last question, is there another method to recover a faulty grasshopper? I have a third grasshopper that is not detected by arduino, it do not appear in the ports section, if I press boot and reset it sound like the board is disconnecting and connecting from windows but it do not appear.