kriswiner/CMWX1ZZABZ #27: Downlink Data Location

meakash, 2020-08-20:
I am using a grasshopper board to send data to TTN whenever i press a button.
The uplink and downlink packets gets incremented everytime i send an uplink packet.

I want to know what is being send as downlink, where it is stored, how to read that downlink message and i know how to use Node Red with Arduino based end nodes to send specific downlink packets, is there any way in which i can send a custom downlink?

Please bear with me as i am just starting with LoRaWAN and grasshopper board.

kriswiner, 2020-08-20:
See the LoRaWAN_Callbacks.ino example:

    if (LoRaWAN.parsePacket())
    {
        uint32_t size;
        uint8_t data[256];

        size = LoRaWAN.read(&data[0], sizeof(data));

        if (size)
        {
            data[size] = '\0';

            Serial.print(", PORT: ");
            Serial.print(LoRaWAN.remotePort());
            Serial.print(", DATA: \"");
            Serial.print((const char*)&data[0]);
            Serial.println("\"");
        }
    }

meakash, 2020-08-21:
Thank you so much sir, this works for me.

meakash, 2020-08-21:
int setReceiveDelay(unsigned int delay);

Sir, i would like to play with delays for a development work (although not correct to do it).
RX2 delay can be increased easily, what about RX1, how to increase it?