kriswiner/CMWX1ZZABZ #30: B-L072Z-LRWAN1 Arduino Sketch Uploading Error

ChaseYoDream, 2020-12-14:
I'm having issues getting the Arduino sketch to upload.  I followed the instructions and my device shows up as COM( B-L072Z-LRWAN1). Here is the error report I'm seeing: 
***************************************************************************************************************
GNU MCU Eclipse 32-bits Open On-Chip Debugger 0.10.0+dev-00487-gaf359c18 (2018-05-12-15:41)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
srst_only separate srst_nogate srst_open_drain connect_assert_srst
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Info : STLINK v2 JTAG v28 API v2 SWIM v18 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.237647
Info : STM32L072.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : Listening on port 3333 for gdb connections
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
adapter speed: 240 kHz
Error: timed out while waiting for target halted

the selected serial port 
*******************************************************************************************************
Any help would be appreciated.

kriswiner, 2020-12-14:
Is your board using the CMWX1ZZABZ-078 module? Did you put the board into
boot mode before trying to flash?

ChaseYoDream, 2020-12-14:
It uses the CMWX1ZZABZ-091 module. I am not sure how to put this dev board into boot mode. I read something about toggling reset while pressing down boot. The only other button is User.

kriswiner, 2020-12-14:
Yes, you can hold boot to GND while powering on or hold boot to GND and
press reset. But it needs to be in boot mode to accept new programming as
far as I know.

Maybe switch to this
<https://www.tindie.com/products/tleracorp/grasshopper-loralorawan-development-board/>
?

ChaseYoDream, 2020-12-15:
No luck with holding Boot0 to GND, and powering up or pressing RESET. I have two of these DEV boards free from work so naturally I'm going to try them out.

kriswiner, 2020-12-15:
No idea, maybe ask at the ST forum then?

kriswiner, 2022-07-18:
> debug_level: 0
> Error: libusb_open() failed with LIBUSB_ERROR_NOT_FOUND
> Error: open failed
> in procedure 'program'
> in procedure 'init' called at file "embedded:startup.tcl", line 496
> in procedure 'ocd_bouncer'
> ** OpenOCD init failed **
> shutdown command invoked
> does not exist or your board is not connected
> How to fix

What board is it? What board variant are you using? Via SWD or USB?

ProjectEIEI, 2022-07-18:
> What board is it? What board variant are you using? Via SWD or USB?
> […](#)

B-L072Z-LRWAN1
STM32L0 Discovery kit LoRa, Sigfox, low-power wireless and USB